dgomes/dali2mqtt #1: No errors though no light activity

beireken, 2019-02-18:
I have installed the library, not using venv (didn't get it to work that way) and it seems to run fine.
I configured it with 1 light in config.ini.
Home assistant sees the MQTT lights but when i switch light 0 it does not do anything.

When i attach the hasseb adapter to a windows machine i can switch the ballast on and off using the stock hasseb software.
Can i debug this in any way? Or is there more config to do then just setting the mqtt broker in config?

I am running in a virtual environment, but the windows machine is too. Further i am running hass.io.
Dali2mqtt is runing on a ubuntu virtual machine.

`dali_driver: hasseb
dali_lamps: 1
ha_discover_prefix: homeassistant
mqtt_base_topic: dali2mqtt
mqtt_port: 1883
mqtt_server: 192.168.5.120
`

dgomes, 2019-02-18:
Please edit https://github.com/dgomes/dali2mqtt/blob/6901988ec7de8360b8d6e586298784f514cf1c2f/dali-mqtt-daemon.py#L58

and change INFO into DEBUG

run the daemon from the command line (not systemd) and share the debug results.

beireken, 2019-02-19:
Ok, my ballast has address 1.
This is me toggling all available dali lamps in HA.

`2019-02-19 06:41:34,114 DEBUG: Loading configuration from <config.ini>
2019-02-19 06:41:34,115 DEBUG: Using <hasseb> driver
2019-02-19 06:41:34,132 DEBUG: Connecting to 192.168.5.120:1883
2019-02-19 06:41:34,134 DEBUG: Search for Lamp 0
2019-02-19 06:41:34,240 DEBUG: No Answer
2019-02-19 06:41:34,369 DEBUG: No Answer
2019-02-19 06:41:34,529 DEBUG: No Answer
2019-02-19 06:41:34,721 DEBUG: No Answer
2019-02-19 06:41:35,008 DEBUG: No Answer
2019-02-19 06:41:35,456 DEBUG: No Answer
2019-02-19 06:41:36,224 DEBUG: No Answer
2019-02-19 06:41:37,569 WARNING: 0 not present:
2019-02-19 06:41:37,570 DEBUG: Search for Lamp 1
2019-02-19 06:41:37,697 DEBUG: No Answer
2019-02-19 06:41:37,824 DEBUG: No Answer
2019-02-19 06:41:37,985 DEBUG: No Answer
2019-02-19 06:41:38,177 DEBUG: Search for Lamp 2
2019-02-19 06:41:38,304 DEBUG: Search for Lamp 3
2019-02-19 06:41:38,560 DEBUG: No Answer
2019-02-19 06:41:38,689 DEBUG: No Answer
2019-02-19 06:41:38,849 DEBUG: QueryActualLevel = BackwardFrame(8,[254])
2019-02-19 06:41:38,976 DEBUG: QueryActualLevel = BackwardFrame(8,[254])
2019-02-19 06:41:39,104 DEBUG: QueryActualLevel = BackwardFrame(8,[254])
2019-02-19 06:41:58,605 DEBUG: Command on dali2mqtt/1/light/switch: b'OFF'
2019-02-19 06:41:58,605 DEBUG: Set light <1> to b'OFF'
2019-02-19 06:41:59,700 DEBUG: Command on dali2mqtt/2/light/switch: b'OFF'
2019-02-19 06:41:59,701 DEBUG: Set light <2> to b'OFF'
2019-02-19 06:42:02,425 DEBUG: Command on dali2mqtt/3/light/switch: b'OFF'
2019-02-19 06:42:02,425 DEBUG: Set light <3> to b'OFF'
2019-02-19 06:42:06,640 DEBUG: Brightness Command on dali2mqtt/1/light/brightness/set: b'254'
2019-02-19 06:42:06,640 DEBUG: Set light <1> brightness to 254
2019-02-19 06:42:07,669 DEBUG: Brightness Command on dali2mqtt/2/light/brightness/set: b'254'
2019-02-19 06:42:07,670 DEBUG: Set light <2> brightness to 254
2019-02-19 06:42:11,141 DEBUG: Brightness Command on dali2mqtt/3/light/brightness/set: b'254'
2019-02-19 06:42:11,141 DEBUG: Set light <3> brightness to 254
2019-02-19 06:42:12,479 DEBUG: Brightness Command on dali2mqtt/0/light/brightness/set: b'254'
2019-02-19 06:42:12,479 DEBUG: Set light <0> brightness to 254
2019-02-19 06:42:14,973 DEBUG: Command on dali2mqtt/0/light/switch: b'OFF'
2019-02-19 06:42:14,973 DEBUG: Set light <0> to b'OFF'
2019-02-19 06:42:17,938 DEBUG: Command on dali2mqtt/1/light/switch: b'OFF'
2019-02-19 06:42:17,938 DEBUG: Set light <1> to b'OFF'
2019-02-19 06:42:18,449 DEBUG: Command on dali2mqtt/2/light/switch: b'OFF'
2019-02-19 06:42:18,449 DEBUG: Set light <2> to b'OFF'
2019-02-19 06:42:19,006 DEBUG: Command on dali2mqtt/3/light/switch: b'OFF'
2019-02-19 06:42:19,006 DEBUG: Set light <3> to b'OFF'
2019-02-19 06:42:21,206 DEBUG: Brightness Command on dali2mqtt/2/light/brightness/set: b'121'
2019-02-19 06:42:21,206 DEBUG: Set light <2> brightness to 121
2019-02-19 06:42:31,677 DEBUG: Brightness Command on dali2mqtt/2/light/brightness/set: b'254'
2019-02-19 06:42:31,677 DEBUG: Set light <2> brightness to 254
`

beireken, 2019-02-19:
I have now quickly set up a pi. With this I succeed in switching the lights.
I do notice that the standard user does not have permissions to run libusb device.
I have to run manually as root. Your stock config uses the user "homeassistant" , did you also experience this?

Also, can we change the fade in/out times of a ballast when switched on/off?

dgomes, 2019-02-19:
Could thing you spotted that :) now I can update the documentation. The user running the daemon must have permission to access the device.

Yes it should be possible, but it's not implemented. Will accept a Pull Request with that :)

beireken, 2019-02-19:
Ok, I guess I should use udev rules for the permissions? It's not working now but I'll figure that out eventually.

I really wish I could be of any help on implementing stuff but I have no experience whatsoever in coding :) 
Is fade in/out a fixed value in the ballasts? Maybe I can set it with hadden config tool?

dgomes, 2019-02-19:
You can set a default fade in/out value or make it dynamic. So yes :)

beireken, 2019-02-19:
Thx for the info!

Is adding to dialout enough? The pi is by default a member of dialout it seems.
Still get:

usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

dgomes, 2019-02-19:
Sorry, the hasseb is not a serial device... (also run a serial2mqtt daemon on the same machine)

I've now updated the README.md, please test

beireken, 2019-02-19:
Yes, using the udev rule works now, thx!

When using the systemd service it does not switch the lights, running manually works fine. I'll have to look into that further.

dgomes, 2019-02-19:
you might need to edit the systems conf file

beireken, 2019-02-20:
I'm closing this issue. Somehow this evening it started working using the systemd service.
I'll have to evaluate how stable it will be. I have no clue why it started working...

closed 2019-02-20 by beireken