dgomes/dali2mqtt #2: Dali_lamps setting ignored?

beireken, 2019-02-20:
In config.ini ( the readme says config.yaml btw?) the dali_lamps setting is default to 4.
When i change this to 2 i would expect the initial searching to stop at lamp 1 ? 
It just continues to lamp 3 though, resulting in 4 entries in my home-assistant set-up  with 2 that don't exist actually.

dgomes, 2019-02-20:
Again thank you for your debug :)

I have 4 ballast (reason why I it defaults to 4) and never tested less then those :(

I've fixed the code and the README!

beireken, 2019-02-20:
Thank you. Works fine now!
I also did a reboot now, the pi came up with the daemon working as it should.

I'll test some more, but could it be that when it scans for lamps that aren't connected it hangs somewhere?

closed 2019-02-20 by beireken

dgomes, 2019-02-20:
It should not hang, I've setup a 30 seconds timeout if it doesn't find a lamp