dgomes/dali2mqtt #5: Dali groups

beireken, 2019-07-30:
Hello Diogo,

Apparently python-dali supports groups, are there plans to inegrate this into dali2mqtt?

https://github.com/sde1000/python-dali/issues/55

I have created a couple HA grouping, but there is always a slight delay between them, using dali groups with same lamps this does not occur.
And just to let you know, for the past months, dali2mqtt has been working flawlessly!

dgomes, 2019-07-31:
Hi

Personally no plans or time :(

But if you do it and make a pull request I will gladly integrate and support.

beireken, 2019-07-31:
Ok, that is way above my skill level :)

Am i right in thinking i could use manual mqtt light config in HA, and add the mqtt callback on the topics for the group  commands in dali2mqtt?
That would be easier than adding automatic discovery.

Tiitan, 2020-08-09:
Hi here, it seems that Ogerman added group support on his fork: https://github.com/ogerman/dali2mqtt

dgomes, 2020-08-09:
> Hi here, it seems that Ogerman added group support on his fork: https://github.com/ogerman/dali2mqtt
> edit: never-mind, it is just logging info yet

That is awesome! I would gladly accept a PR

crami, 2022-01-07:
I have contributed group support a few month ago, have not seen this issue. Please try with the current code.

crami, 2022-01-07:
Sorry, pull request is nor accepted yet, you can get my code at: https://github.com/crami/dali2mqtt/tree/dali-groups

dgomes, 2022-01-09:
8c6e4ef74d4072babb45ac9f1fa6c047756d2f5d closes this issue :)

closed 2022-01-09 by dgomes