dgomes/dali2mqtt #7: Color light and multiple busses support ?

Tiitan, 2020-08-09:
Hi, 

I'm new to dali so I could write some nonsense here.

it does not seems that color light (using multiple channel) are supported , is it?
if not could, it be supported directly from here or should it be supported by python-dali first ? Seems that a layer of abstraction is lacking between a light representation and a dali channel. multiple light type (cct, rgb, rgbw, rgbcct) covering different range of channels with a dedicated payload parsing for each would be necessary.
I'm also wondering if the dali protocol altogether can tell when a single balast uses multiples channels. if not proper device discovery is impossible and the device map would need to come from a config file.

also, I believe I will largely exceed 64 channel on my setup so I need to separate my house into multiple dali busses and connect several dali-usb controllers to my server. Could this plugin (and python-dali) manage multiple dali busses ?

dgomes, 2020-08-09:
Hi Titan,

This project is only about exposing https://github.com/sde1000/python-dali through MQTT.

I think your question are pertinent, but I don't have answers as I only have 4 lights controlled through DALI :)

You should direct these question to python-dali

Tiitan, 2020-08-09:
I only saw DT6 multi-channel controllers for sale, so I assumed it was standard for color light. but as I keep investigating dali, I discovered that the DT8 standard is now "half" certified as of february 2020 and it allow color lights to use a single channel.
 
- there is already an issue about that implementation on python-dali: https://github.com/sde1000/python-dali/issues/52
- If I can get my hands on these new controlers I guess I wont need multiple busses after all

closed 2020-08-09 by Tiitan