dgomes/dali2mqtt #14: Send DALI devices as separate MQTT Devices

AdamN1, 2020-08-31:
Hi,
At the moment each DALI devices is send with one [MQTT device](https://www.home-assistant.io/integrations/light.mqtt/#device) setting. Configuration causes that DALI devices are discovered separate entities but one device by HA.

This is limitation because actual version(1.14.x) of HA can't assign area to entities only to devices.

My proposition: adding configuration option to choose how DALI devices are exported via MQTT to HA.

@dgomes what you about this?
If you are ok with this we(@JustNet) can prepare patch with PR.

dgomes, 2020-08-31:
This issue is similar to HUE, I say we wait for a generic solution coming from HA before addressing this in dali2mqtt.

There have already been some discussions about it in the HA developers channel.

AdamN1, 2020-08-31:
I know that HA is working(thinking) about expanding areas support but is there a deadline scheduled?
If there is no deadline for adding areas entity support let's support this in the dali2mqtt bridge by configuration switch. I think it will not harm anyone.

If you are not rejcecting proposed option I'm willing to make a patch.

dgomes, 2020-08-31:
Don't think it is good to shoehorn that functionality, the device is the gateway not the lamps.

Let's leave this open until HA progresses

AdamN1, 2020-10-10:
I installed zigbee2mqtt and saw that attached device to zigbee module is added to HA as device not entity. Let's reconsider one more time supporting this option.

dgomes, 2022-04-17:
Home assistant now enables you to change area of a specific entity

closed 2022-04-17 by dgomes

AdamN1, 2022-04-18:
> Home assistant now enables you to change area of a specific entity

Yes this works greate.