dgomes/dali2mqtt #15: Don't overwrite config file after each run

AdamN1, 2020-08-31:
Script when closing alway overwrite config file with setting fetched on start. 
This create situation where we can't change configuration without exiting script. Better aproache is to read file at startut, and don't override on exiting.

dgomes, 2020-08-31:
this behaviour is very similar to zigbee2mqtt, my intention was to eventually provide MQTT topics to change the configuration on the fly...

AdamN1, 2020-08-31:
> this behaviour is very similar to zigbee2mqtt, my intention was to eventually provide MQTT topics to change the configuration on the fly...

Will HA have support for this or do you like to change configuration from separate mqtt client?

dgomes, 2020-08-31:
> Will HA have support for this or do you like to change configuration from separate mqtt client?

you can have both way :)

AdamN1, 2020-09-01:
I come with proposition to resolve the case:
1. script don't overwrite configuration on closing
2. MQTT, introduction of new functions to work with config:
   - mqtt_config_read(variable) - will return config variable or all variable when variable is empty
   - mqtt_config_set(variable,value) - will set variable to value
   - mqtt_config_commit - will commit changes to config file

In separate ticket we can add check if the config file didn't change after first run. If file was changed we block commit of config variable. 

@dgomes what you think?

dgomes, 2020-09-01:
you don't need those many functions... I'll put in a PR with an example of what makes since to me

AdamN1, 2020-09-05:
> you don't need those many functions... I'll put in a PR with an example of what makes since to me

@dgomes Can you give some example so we can start working on this?

dgomes, 2020-09-05:
I've already started a new configuration class to handle this, hope to finish it soon

dgomes, 2020-09-07:
@kbacia7 please test and check if we can close this

kbacia7, 2020-09-07:
Tested on newest `master`, works

closed 2020-09-07 by dgomes