dgomes/dali2mqtt #16: Adding color support for the system log

AdamN1, 2020-08-31:
Adding color support for the system log like:

- error - red
- notice - default
- information status (ok) - green
- information status (bad) - red

dgomes, 2020-08-31:
as long as it still uses python logging, is optional and off by default I'm OK

AdamN1, 2020-09-02:
> as long as it still uses python logging, is optional and off by default I'm OK

@dgomes Is there a way to convince you to turn on by default log coloring(red) for 'errors' ?
I think today all terminals support coloring so it shouldn't be a issue.

AdamN1, 2020-09-02:
I only want to color console output on the terminal.
Maybe we can implement something like -F --foreground with color on?

dgomes, 2020-09-02:
that is acceptable

(please use separate PR's for each issue)

dgomes, 2020-09-03:
closed by #22

closed 2020-09-03 by dgomes