dgomes/dali2mqtt #17: Fix max lamp implementation vs DALI range

AdamN1, 2020-08-31:
Hi,
In script we can set dali-lamps number but latter this map to max_range which is responsible for limiting the addresses to be searched. Myślę, że implementacja jest błędna. Załóżmy sytuację z czterema lampami:

1. lamp 1 address 0
2. lamp 2 address 2
3. lamp 3 address 4
4. lamp 4 address 6
In the current implementation script will find only the first two lamps.

I think that we need to separate number of lamps from dali address range. This will allow people to use non continuous address number for the lamps.

dgomes, 2020-08-31:
(please translate)

AdamN1, 2020-08-31:
Fixed :D

AdamN1, 2020-09-03:
@dgomes I think that PR #21 should be rejected and let's start second  new clean approach on new PR.

I think PR should address the problems:

1. adding support for an address range (dali short address) to be searched, what you think about variable name: --dali-shortaddress-range -dsr?
2. stop search when --dali-lamps is meet, what you think about rename this to --dali_lamps_number -dln?

AD.1 variable --dali-shortaddress-range -dsr should support:

- range like 5-12 - search short address from 5 to 12
- name full - full search of dali range from 0 to 63, default option

AdamN1, 2020-09-05:
@dgomes are you ok with my proposal? If you are ok, we can start work on this.

dgomes, 2020-09-05:
Actually since the new firmware was release, this initial scan (which used to take several seconds) is now very fast. 

Maybe we can simply remove this option and always scan all the addresses ?

AdamN1, 2020-09-06:
I don't see any harm in adding this. If someone want to scan only part of dali short range let's give them this possibility.
If scanning speed isn't issue now, let's make default to scan full range, ok?

dgomes, 2020-09-06:
Lets go full range!

AdamN1, 2020-09-06:
> Lets go full range!

So we delete **dali-lamps** and set range 0,63 without options to customize, yes?

closed 2020-09-07 by dgomes