dgomes/dali2mqtt #30: Support for RGB control of DT8 device type

AdamN1, 2020-09-06:
Anyone with DALI RGB devices so we can test implement this and test?

Ipallis, 2020-09-14:
I have a DT8 device but not a RGB one. It is a tunable white device and luminates from 2700K (warm white) to cold White (5000K) using the `color_temp` attribute in mireds of  Light integration. If you are interested to support that, i would add it in the next time.

dgomes, 2020-09-14:
@Ipallis a PR will be welcomed!

AdamN1, 2020-09-14:
> I have a DT8 device but not a RGB one. It is a tunable white device and luminates from 2700K (warm white) to cold White (5000K) using the `color_temp` attribute in mireds of Light integration. If you are interested to support that, i would add it in the next time.

Can you write power supply full model name?

ogerman, 2023-04-20:
#https://github.com/sde1000/python-dali/issues/52

rohanramaswamy, 2024-08-17:
I have a working proof of concept of DT8 color temperature control working well with HA here:
https://github.com/rohanramaswamy/dali2mqtt
I would love feedback so I can move towards a PR as I have made some radical changes.

dgomes, 2024-08-18:
@rohanramaswamy I would gladly accept it if you split it in various PR:
- Change to Async
- Add color temperature control
- And any other features

Asking to split in order to ease the process of validating the changes in various users setups.