dgomes/dali2mqtt #41: Remove lamps from MQTT on exit

kbacia7, 2020-10-21:
I think that good idea will be removing all lamps from MQTT on exit of script. Currently after a few hours of experincing we can have many lamps on MQTT which doesn't exists

dgomes, 2020-10-21:
That is not so great idea... it would trigger event in supporting systems such as Home Assistant.

Its normal for this to happen in development, not so much in production

closed 2020-10-21 by dgomes

kbacia7, 2020-10-21:
But after friendly names, If I change name of lamp in `devices.yaml`, on MQTT will be new name and old

dgomes, 2020-10-21:
just like in zigbee2mqtt, you delete the old topic yourself if you change the name in the configuration file (by hand)

Or if you change using an MQTT message, you then delete the old topic.

To delete everything on exit is not a good idea, a simple reboot triggers changes in HA (not advisable)