dgomes/dali2mqtt #43: Map HA dali brightness level to DALI min-max range

AdamN1, 2020-11-18:
After we query lamps about supported brightness range we can create map of home assistant brightness level to dali range.
My proposition:

1. HA 100% - Dali level 254, 255 is reserved
2. HA from 99% to 1% - map from Dali 254 to physical minimum( received from DALI device ) or default 85(85 is default for tridonic devices) 
3) HA 0% - DALI lamp off

ad.2 calculation of DALI brightness steps steps, (254(max) - 85(min))/99 next round the number up 

Above proposal solves the problem of the inability to set the brightness below 34% for lamps with Minimal brightess level like 85 and similar

AdamN1, 2021-05-07:
> I think we should keep as it is and just turn off the light if brightness is set bellow the threshold, in recent HA versions we can send attribute values, so we can inform the user of the threshold.

I think that we can only set `maximum brightness` not the minimal or I'm missing something?

> First there is no need to change from 0..255 to 0..100, HA handles 0.255 natively.

I'm not thinking about this translation but translation from 85 to 255. '85' is the minimum supported brightness that represent value 34 in brightness bar in the HA UI.
[image]

If i change bar to smaller value this not change anything with light value. I don't think this behavior is intuitive.I know that best approach is to solve the problem in the HA, unfortunately no one is working on this. So maybe we can fix this by mapping values in the dali2mqtt?

AdamN1, 2021-05-07:
> My point is that doing this mapping in dali2mqtt will not produce a good result, as there are various mappings and rounding going in dali -> mqtt -> HA backend -> HA frontend
> 
> We should instead propose a MQTT light configuration option in HA that can be used to set the minimum level.

I agree this is best approach, time to read HA devel docs :(

dgomes, 2022-04-17:
As of today HA considers 0 as off and minimum is 1

closed 2022-04-17 by dgomes

AdamN1, 2022-04-18:
> As of today HA considers 0 as off and minimum is 1

How this solve our problem?

dgomes, 2022-04-18:
At the time of this issue you could set brightness to 0, that is no longer possible minimum brightness is 1 (0 is OFF in DALI)