dgomes/dali2mqtt #47: Lamp delation

AdamN1, 2021-05-07:


AdamN1, 2022-04-17:
@dgomes ??

dgomes, 2022-04-17:
and how will you distinguish those ? 

I think we should move away from retained topics and publish the topics upon home assistant restart

AdamN1, 2022-04-18:
> and how will you distinguish those ?

We can get list from mqtt server and compare records in topic.

> I think we should move away from retained topics and publish the topics upon home assistant restart

I don't think that I understand what you mean

dgomes, 2022-04-18:
discovery topics need to be there only during HA start, you don't need to "store" them in the MQTT broker using the retain flag. 

Each time HA starts it already publishes a message, dali2mqtt should subscribe to that message and publish all the lamps without retain flag, in this way there is no need to clear topics as they are never retained.