dgomes/dali2mqtt #51: No MQTT Topics

pafra, 2021-09-07:
Hi all,

I have installed dali2mqtt according to your instructions.

But on the MQTT server I don't see any topics from the lamps.
But I see the status of the dali2mqtt.

What could be the problem?

Here are the output:
```bash
~/dali2mqtt-venv/bin/python3 ~/dali2mqtt/dali_mqtt_daemon.py --config ~/dali2mqtt/config.yaml --log-level debug --log-color 
2021-09-07 16:37:16,567 DEBUG: Loading devices names from <devices.yaml>
2021-09-07 16:37:16,574 DEBUG: Using <tridonic> driver
2021-09-07 16:37:16,666 DEBUG: Connecting to 192.168.4.201:1883
2021-09-07 16:37:16,681 DEBUG: Loading devices names from <devices.yaml>
2021-09-07 16:37:18,566 INFO: Found 2 lamps
```

```bash
mosquitto_sub -h 192.168.4.201 -u myuser -P mypassword -t dali/+ -v
dali/status offline
dali/status online
```

thx

dgomes, 2021-09-07:
can you share you config.yaml ?

pafra, 2021-09-07:
The device was removed at the start of `dali_mqtt_daemon.py`

There is no massage int the log or `dmesg`

```bash
ls -l /dev/hidraw0
crw------- 1 root root 237, 0 Jul 21 19:00 /dev/hidraw0

# Start daemon
~/dali2mqtt-venv/bin/python3 ~/dali2mqtt/dali_mqtt_daemon.py --config ~/dali2mqtt/config.yaml --log-level debug --log-color 
2021-09-07 16:37:16,567 DEBUG: Loading devices names from <devices.yaml>
2021-09-07 16:37:16,574 DEBUG: Using <tridonic> driver
2021-09-07 16:37:16,666 DEBUG: Connecting to 192.168.4.201:1883
2021-09-07 16:37:16,681 DEBUG: Loading devices names from <devices.yaml>
2021-09-07 16:37:18,566 INFO: Found 2 lamps

# and then the device is gone
ls -l /dev/hidraw0 /dev/usb/hiddev0                                                                   
ls: cannot access '/dev/hidraw0': No such file or directory
```

pafra, 2021-09-07:
Yes, 

`cat config.yaml `
```yaml
dali_driver: tridonic
devices_names: devices.yaml
ha_discovery_prefix: homeassistant
log_color: true
log_level: debug
mqtt_base_topic: dali
mqtt_password: mypassword
mqtt_port: 1883
mqtt_server: 192.168.4.201
mqtt_username: myuser
```

pafra, 2021-09-07:
I am using Ubuntu 20.04.3 aarch64 (Raspi)

dgomes, 2021-09-07:
strange.... I don't have a tridonic device myself to test :(

pafra, 2021-09-23:
@dgomes what OS do you using?

dgomes, 2021-09-23:
I'm running dali2mqtt on the latests raspian version

pafra, 2021-09-23:
thx, I will test it in the next days

AdamN1, 2022-04-17:
> strange.... I don't have a tridonic device myself to test :(
> 
> @AdamN1 do you have a tridonic ?

Unfortunately not, i will try search something and test.