dgomes/dali2mqtt #53: Periodic lamp status reports

crami, 2022-01-07:
As I do have native dali light switches which also control the lights i do not get the "real" status over mqtt. It would be nice if we could implement periodic status messages.

I think we have to do `driver_object.send(gear.QueryActualLevel(short_address))` in periodic intervals (all 5-10s) for every lamp and emit a mqtt message with the result.

Looking at the code I currently have no idea where to implement this in a nice way. If there are any ideas please post here so I can try to implement this. Or send me code :-)

dgomes, 2022-01-07:
We can replace the `loop_forever()` with 
```
while run:
    mqttc.loop()
    #get DALI states
 ```

crami, 2022-01-08:
Yes, was at that point too. What I do not get is how I can access the data_object from a function called there. I followed it down the rabit hole until this:

`mqttc.on_connect = lambda a, b, c, d: on_connect(a, b, c, d, ha_prefix)`

and there I get lost.
If you have a hint to that that would be great.

crami, 2022-01-08:
Just implemented a get command in my fork, so I can poll the brightness levels, that is probably the cleaner way as not all users need this.

closed 2022-01-08 by crami

dgomes, 2022-01-08:
I think this makes sense to be included. Feel free to upon the PR :)