dgomes/dali2mqtt #59: WIP: New installation - master branch fixes

AdamN1, 2022-04-17:
Hi,
I tride prepare new instalation of dali2mqtt using manual from readme and meet couple problems:

1. missing information about system requirement, I propse new step in documentation 
**Install the following dependacy's with this command:**
```
apt-get install \
git \
python3-pip \
python3-venv \
libhidapi-dev \
```
2. Move python dependency wheel from requirements.txt to **pip install wheel** becouse when it's installed from requirements other modules use installation in legacy mode.
```
Using legacy 'setup.py install' for paho-mqtt, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pathtools, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pyusb, since package 'wheel' is not installed.
Using legacy 'setup.py install' for voluptuous, since package 'wheel' is not installed.
Using legacy 'setup.py install' for watchdog, since package 'wheel' is not installed.
Using legacy 'setup.py install' for python-slugify, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pyhidapi, since package 'wheel' is not installed.
Using legacy 'setup.py install' for slugify, since package 'wheel' is not installed.
```
3. Missing option in default config.yaml created after first run. Second run of the scripts return error
```
mqtt_username: zigbee2mqtt
mqtt_password: losowehaslodozigbee
```

dgomes, 2022-04-17:
1. has information that is too much distribution specific
2. Don't understand what you are proposing
3. username and password are optional

dgomes, 2022-04-17:
actually started an upgrade rampage :)

I depend on https://github.com/sde1000/python-dali/pull/99 but this PR will move us towards a more streamline approach to install without any strange dependencies.

3. fixed in `master`

AdamN1, 2022-04-18:
> See https://github.com/dgomes/dali2mqtt/blob/master/Debian_Ubuntu.txt

@dgomes can you link this to README(make section). 

Maybe we can change naming scheme? Debian, Ubuntu user distribution suffix like Readme.Debian, changelog.Debian.
So according to Debian name schema name should be README.Debian.md

What about other dependency, they aren't normalny installed with the system. I know that user should know what they are doing but adding aditional package(git ...) will help in the futere get rid of simple instalation question about missing commands.

dgomes, 2022-04-18:
I started a fresh development environment and all was necessary was python3-hid (whose dependencies pulled everything else). I'll move into a README subsection since it's soo small.

dgomes, 2022-04-21:
just merged an update on requirements.txt

New version should solve dependencies problems :)

dgomes, 2022-04-24:
I think this is all sorted, also doing extra housekeeping ;)

closed 2022-04-24 by dgomes