dgomes/dali2mqtt #62: Installation?

VoetenK, 2023-11-06:
I have made many attempts to install the dali2mqtt.
I want to install it on a Raspberry Pi 4B with Linux Debian 12 (Bookworm) as the operating system.
I am working directly on my Raspberry Pi. (screen, keyboard) under the username "homeassistant". The Hasseb USB DALI Master in a USB port.
I do the installation in the folder home/homeassistant/.
When I run the instruction "venv/bin/python3 -m dali2mqtt.dali2mqtt", I get as a result the configuration of my DALI devices. I complete the rest of the installation.
The problem is that when I control or turn on or off my lights, I do not receive any messages in homeassistant. 
Can you possibly list step by step how I should do the installation? Or can you explain the problem?

dgomes, 2023-11-06:
What about on the MQTT broker ? Do you see the messages being published there ?

VoetenK, 2023-11-06:
Thanks for the quick response.
Before that I installed mosquitto-clients. ( sudo apt-get install mosquitto-clients)
This shows online and when I send a message (topic: "dali2mqtt" payload: 75") in Node Red it arrives at the terminal. In Node Red I also get a reply back .

Terminal:
homeassistant@debianha:~/dali2mqtt$ mosquitto_sub -L mqtt://mqttuser:******@192.168.0.8:1883/dali2mqtt/#       
online
75

Node Red:
6/11/2023 19:09:55node:445dcbdea289ef13
dali2mqtt : msg : Object
object
_msgid: "ee5cbdeb24e112d5"
topic: "dali2mqtt"
payload: "75"
6/11/2023 19:09:55node: fbc3be5d452a8232
dali2mqtt : msg : Object
object
topic: "dali2mqtt"
payload: "75"
qos: 2
retain: false
_msgid: "0169130a4d88f900"

So far it does work??

dgomes, 2023-11-06:
OK and you have MQTT integration in HA configured properly ?

VoetenK, 2023-11-06:
As far as I can judge, yes. I use Mosquitto broker in Home Assistant to use https://github.com/Chibald/maestrogateway. This one works, but.

homeassistant@debianha:/$ mosquitto_sub -v -t "#" -u mqttuser -P ******
homeassistant/status offline
dali2mqtt/status online

$ homeassistant@debianha:/$ mosquitto_sub -h 192.168.0.8 -u mqttuser -P ******* -t dali2mqtt/# -v
dali2mqtt/status online

$ homeassistant@debianha:/$ mosquitto_sub -h 192.168.0.8 -u mqttuser -P ******* -t homeassistant/light/# -v
Connection error: Connection Refused: not authorized.

How can I resolve this?

dgomes, 2023-11-06:
pretty use you should not have "not authorised"... 

but homeassistant looks like it's not connected to MQTT Broker :/ ?

VoetenK, 2023-11-06:
homeassistant@debianha:~$ sudo systemctl enable homeassistant.service
Failed to enable unit: Unit file homeassistant.service does not exist.

I don't understand what's wrong.
Can you explain to me how to fix this?

dgomes, 2023-11-06:
please follow https://www.home-assistant.io/integrations/mqtt/

VoetenK, 2023-11-06:
my settings:

Opties
Logins
- username: mqttuser
  password: ******

Certificate File
fullchain.pem

Private Key File
privkey.pem

active: false
folder: mosquitto

1883
1883/tcp
Normal MQTT
1884
1884/tcp
MQTT over WebSocket
8883
8883/tcp
Normal MQTT with SSL
8884
8884/tcp
MQTT over WebSocket with SSL

dgomes, 2023-11-06:
that's probably your HA Add-on configuration... besides that you need to actually configure HA to use it (follow the documentation)

This is getting out of topic :) you better resort to community.home-assistant.io

VoetenK, 2023-11-08:
I started from scratch with a clean Linux operating system. Attached you will find the full course of dali2mqtt installation.
mss you can find a problem in this?

I still have the problem that when I turn off or dim a lamp I receive no message in my mqtt client.
Messages sent with Node Red do pass.

Do you have any sugessties?
[dali2mqtt.txt](https://github.com/dgomes/dali2mqtt/files/13299056/dali2mqtt.txt)

dgomes, 2023-11-08:
can you execute dali2mqtt with debug messages ?

`$ LOGLEVEL="DEBUG" venv/bin/python3 -m dali2mqtt.dali2mqtt`

VoetenK, 2023-11-08:
root@debianha:/home/homeassistant/dali2mqtt# LOGLEVEL="DEBUG" venv/bin/python3 -m dali2mqtt.dali2mqtt
2023-11-08 17:30:29,428 INFO: Found 2 lamps
2023-11-08 17:30:30,122 INFO: kitchen - address: 0, actual brightness level: 74 (minimum: 58, max: 254, physical mini                       mum: 58)
2023-11-08 17:30:30,763 INFO: bathroom - address: 1, actual brightness level: 74 (minimum: 58, max: 254, physical min                       imum: 58)
2023-11-08 17:30:30,764 INFO: Scanning for groups
2023-11-08 17:30:31,381 INFO: Finished scanning for groups
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 687, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 375, in getMessage
    msg = str(self.msg)
          ^^^^^^^^^^^^^
  File "/home/homeassistant/dali2mqtt/dali2mqtt/lamp.py", line 108, in __str__
    f"{self.device_name} - address: {self.short_address.address}, "
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Group' object has no attribute 'address'
Call stack:
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 536, in <module>
    main(args)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 476, in main
    mqttc.loop_forever()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_fore                       ver
    rc = self._loop(timeout)
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_r                       ead
    rc = self._packet_handle()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3039, in _packet_h                       andle
    return self._handle_connack()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3138, in _handle_c                       onnack
    on_connect(
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 397, in <lambda>
    mqttc.on_connect = lambda a, b, c, d: on_connect(a, b, c, d, ha_prefix)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 367, in on_connect
    initialize_lamps(data_object, client)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 210, in initialize_lamps
    create_mqtt_lamp(group_address, f"group_{group}")
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 191, in create_mqtt_lamp
    logger.info(lamp_object)
Message: <dali2mqtt.lamp.Lamp object at 0xffffaa66bcd0>
Arguments: ()
2023-11-08 17:30:32,081 INFO: initialize_lamps finished
^CTraceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 536, in <module>
    main(args)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 476, in main
    mqttc.loop_forever()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1150, in _loop
    socklist = select.select(rlist, wlist, [], timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

root@debianha:/home/homeassistant/dali2mqtt#

dgomes, 2023-11-08:
Well something did not go as expected because I'm missing the DEBUG messages...

Can you activate the vent and run: 

`$ LOGLEVEL="DEBUG" python3 -m dali2mqtt.dali2mqtt `

dgomes, 2023-11-08:
Please stop

VoetenK, 2023-11-08:
(venv) root@debianha:/home/homeassistant/dali2mqtt# sudo systemctl stop dali2mqtt.service
(venv) root@debianha:/home/homeassistant/dali2mqtt# LOGLEVEL="DEBUG" python3 -m dali2mqtt.dali2mqtt
2023-11-08 18:45:44,682 INFO: Found 2 lamps
2023-11-08 18:45:45,376 INFO: kitchen - address: 0, actual brightness level: 58 (minimum: 58, max: 254, physical minimum: 58)
2023-11-08 18:45:46,016 INFO: bathroom - address: 1, actual brightness level: 58 (minimum: 58, max: 254, physical minimum: 58)
2023-11-08 18:45:46,016 INFO: Scanning for groups
2023-11-08 18:45:46,603 INFO: Finished scanning for groups
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 687, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 375, in getMessage
    msg = str(self.msg)
          ^^^^^^^^^^^^^
  File "/home/homeassistant/dali2mqtt/dali2mqtt/lamp.py", line 108, in __str__
    f"{self.device_name} - address: {self.short_address.address}, "
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Group' object has no attribute 'address'
Call stack:
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 536, in <module>
    main(args)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 476, in main
    mqttc.loop_forever()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 397, in <lambda>
    mqttc.on_connect = lambda a, b, c, d: on_connect(a, b, c, d, ha_prefix)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 367, in on_connect
    initialize_lamps(data_object, client)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 210, in initialize_lamps
    create_mqtt_lamp(group_address, f"group_{group}")
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 191, in create_mqtt_lamp
    logger.info(lamp_object)
Message: <dali2mqtt.lamp.Lamp object at 0xffff9b75bd90>
Arguments: ()
2023-11-08 18:45:47,303 INFO: initialize_lamps finished

VoetenK, 2023-11-08:
Every time I run LOGLEVEL="DEBUG" python3 -m dali2mqtt.dali2mqtt I receive a series (+/- 20) of messages on mqtt from Node Red and Homeassistatant integration.

VoetenK, 2023-11-08:
^CTraceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 536, in <module>
    main(args)
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 476, in main
    mqttc.loop_forever()
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/dali2mqtt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1150, in _loop
    socklist = select.select(rlist, wlist, [], timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

dgomes, 2023-11-08:
something is definitely not right... We still don't get DEBUG messages... 

open the file "dali2mqtt.py" and change "INFO" to "DEBUG"

VoetenK, 2023-11-08:
> something is definitely not right... We still don't get DEBUG messages...
> 
> open the file "dali2mqtt.py" and change "INFO" to "DEBUG"

Done that.

(venv) root@debianha:/home/homeassistant/dali2mqtt/dali2mqtt# sudo systemctl stop dali2mqtt.service
(venv) root@debianha:/home/homeassistant/dali2mqtt/dali2mqtt# LOGLEVEL="DEBUG" python3 -m dali2mqtt.dali2mqtt
Traceback (most recent call last):
  File "frozen runpy", line 189, in _run_module_as_main
  File "frozen runpy", line 112, in _get_module_details
  File "/home/homeassistant/dali2mqtt/dali2mqtt/dali2mqtt.py", line 18, in <module>
    from dali2mqtt.devicesnamesconfig import DevicesNamesConfig
ModuleNotFoundError: No module named 'dali2mqtt.devicesnamesconfig'; 'dali2mqtt' is not a package
(venv) root@debianha:/home/homeassistant/dali2mqtt/dali2mqtt#

VoetenK, 2023-11-08:
Okay, I have discovered something. 
My lights at address 0 and 1 appeared as entity. :-) I can switch these on and off and control them here.
It also shows a group with address 0. I cannot control this one.
I created this groups with a Tridonic DALI USB device.
I will first experiment with this.
What I find strange is that I can control the lights with my Tridonic but I see no reaction in Home Assistant.
Am I doing something wrong here.

dgomes, 2023-11-08:
dali2mqtt is crashing... so nothing can happen

Again... no debug messages...

dgomes, 2023-11-08:
BTW... you could replicate DALI groups in HA

(Deleting the DALI Group would probably fix the issue at hand)