dgomes/dali2mqtt #63: ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'

Hadatko, 2024-02-02:
Hi, i am starting to use this project and from begging i am getting errors. This is first one. I cannot provide much information about that because it disappear meantime and now i have different issue. So far related to this one:

i connected this https://svitimprotebe.cz/eshop/dali-usb/ tridonic usb dali converter to my server and connected to it one dali device. Then i created docker container where all dependencies are installed. The error looks like from python-dali module but actually it looks like this project is trying to access values from errors. I would say this rows but cannot confirm as mentioned:
[image]

```docker
FROM alpine
RUN apk add git python3 py3-pip py3-paho-mqtt py3-slugify py3-usb py3-yaml py3-text-unidecode py3-voluptuous py3-watchdog hidapi 
RUN pip3 install python-dali hidapi --break-system-packages
RUN git clone https://github.com/dgomes/dali2mqtt.git && cd dali2mqtt
ENTRYPOINT ["tail", "-f", "/dev/null"]
```
```bash
/dali2mqtt # python3 -m dali2mqtt.dali2mqtt
2024-02-02 19:29:08,209 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:08,209 DEBUG: Using <tridonic> driver
2024-02-02 19:29:08,227 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:08,230 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:08,305 DEBUG: Found lamp at address 2
2024-02-02 19:29:09,883 INFO: Found 1 lamps
2024-02-02 19:29:09,909 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
2024-02-02 19:29:15,910 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:15,913 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:15,972 DEBUG: Found lamp at address 2
2024-02-02 19:29:17,551 INFO: Found 1 lamps
2024-02-02 19:29:17,577 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
2024-02-02 19:29:20,578 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:20,580 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:20,639 DEBUG: Found lamp at address 2
2024-02-02 19:29:22,217 INFO: Found 1 lamps
2024-02-02 19:29:22,243 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
2024-02-02 19:29:31,244 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:31,247 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:31,306 DEBUG: Found lamp at address 2
2024-02-02 19:29:32,884 INFO: Found 1 lamps
```

dgomes, 2024-02-02:
I don't own such device (have an HASSEB controller), so limited testing... 

This daemon relies on python-dali project, and this clearly is a issue associated with that library which I acknowledge could be better handled in this code.

If you are using containers, I advise you to create a volume to store the configuration yaml's

Hadatko, 2024-02-02:
Yes, i prepared volume for config already. If i will be able to reproduce issue i will provide more information.

Hadatko, 2024-02-03:
Traceback:
```bash
2024-02-03 12:49:54,688 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
^CTraceback (most recent call last):
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 476, in main
    mqttc.loop_forever()
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 397, in <lambda>
    mqttc.on_connect = lambda a, b, c, d: on_connect(a, b, c, d, ha_prefix)
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 367, in on_connect
    initialize_lamps(data_object, client)
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 199, in initialize_lamps
    create_mqtt_lamp(
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 145, in create_mqtt_lamp
    lamp_object = Lamp(
                  ^^^^^
  File "/dali2mqtt/dali2mqtt/lamp.py", line 44, in __init__
    ).value
      ^^^^^
AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 536, in <module>
    main(args)
  File "/dali2mqtt/dali2mqtt/dali2mqtt.py", line 482, in main
    time.sleep(random.randint(MIN_BACKOFF_TIME, MAX_BACKOFF_TIME))
KeyboardInterrupt
```