dgomes/dali2mqtt #64: ERROR: TypeError: '<=' not supported between instances of 'int' and 'str'

Hadatko, 2024-02-02:
I made modification to see why i am getting this error
[image]
With result
1
MASK
254
2024-02-02 20:11:24,767 ERROR: TypeError: '<=' not supported between instances of 'int' and 'str'

Hadatko, 2024-02-02:
After this change i am able to control my dali device
```git
diff --git a/dali2mqtt/lamp.py b/dali2mqtt/lamp.py
index 272948d..581c97f 100644
--- a/dali2mqtt/lamp.py
+++ b/dali2mqtt/lamp.py
@@ -90,6 +90,14 @@ class Lamp:
     @level.setter
     def level(self, value):
         """Commit level to ballast."""
+        if not isinstance(value, (int, float)):
+                   logger.warning(f"The value '{value}' is not a number")
+            self.__level = 1
+            return
+        
+        print(self.min_level)
+        print(value)
+        print(self.max_level)
         if not self.min_level <= value <= self.max_level and value != 0:
             raise ValueError
         self.__level = value
```

dgomes, 2024-02-02:
Why are you setting the level of the light to "MASK" ?! 

it should be a number 0..255

Hadatko, 2024-02-02:
I am not doing anything :D It is your code :D I am just debugging it

dgomes, 2024-02-02:
For that debug message to appear, you have published "MASK" to the MQTT topic

Hadatko, 2024-02-02:
Only if homeassistant is doing that without my knowledge which i doubt. Only if that is some handshake issue. But only i did so far was running your code. 
But maybe there can be another issue i will try to confirm. My container name is same as your project. And i am tracking information about containers using some HA integration. So maybe there is some conflict in topics.

dgomes, 2024-02-02:
Did you do any sort of configuration in Home Assistant ? Because the daemon should have configured all the devices for you, without any human intervention.

Hadatko, 2024-02-02:
Hmm that is weird but MASK string issue disappeared too. So confusing. I will let you know if i will notice it once again.
But i think some fix similar i did would be good to be sure data are correct.

Hadatko, 2024-02-02:
No these errors appeared directly after running your python code. Didn't have time to go and do something in HA

dgomes, 2024-02-02:
A warning should be logged indeed, but I advise against setting an arbitrary level. Would you like to open a PR ?

Hadatko, 2024-02-02:
Yes sure. Just to clarify the MASK word always appeared at the beginning when i run code. But if i kept it running and was doing changes using HA all values were correct. But when i closed and run app again, i got MASK word again and then it worked ok again. Sorry for a lot of "again" words. Now the word is not appearing....

Hadatko, 2024-02-02:
Once i will have more time i will try to debug it more why it appear randomly at the beginning.

dgomes, 2024-02-02:
there is no "MASK" in my code 😬

and worst... that setter is called at https://github.com/dgomes/dali2mqtt/blob/master/dali2mqtt/dali2mqtt.py#L276 which does a cast to int... so no string should have ever reached the setter.

dgomes, 2024-02-02:
I'm available at Home Assistant discord with the same handle as here, if you would like to reach out to expedite debug

Hadatko, 2024-02-02:
I appretiate your offer ;) And also how quick are you responding.
Do you think these level  cals may not cause the issue with mask?
[image]

Hadatko, 2024-02-02:
The bottom one doesn't have ".value" postfix call as the upper once

dgomes, 2024-02-02:
variable is different... it's `__level` not `level`

but if the error is coming from that part of the code it means the driver (Tridonic) is storing "MASK" instead of a value..... again the issue is in python-dali... it need to be standardised there IMHO

Hadatko, 2024-02-03:
Not sure if this is related byt found this in python-dali:
[image]

dgomes, 2024-02-03:
Not doing any Colour queries... would find quite odd to ask for brightness and the driver returning colour :/

Hadatko, 2024-02-03:
So we need handle that differently as  "MASK" word is intended 
[image]
[image]

dgomes, 2024-02-03:
Where is that ? (better link code then to attach screenshots...)

Hadatko, 2024-02-03:
https://github.com/sde1000/python-dali/blob/0ebccd7ed46f2a6bb581f1f4c9c8937181f43ea2/dali/gear/general.py#L796

Hadatko, 2024-11-09:
@hichemsystem1 Hi, i guess you are trying to reach owner.

dgomes, 2024-11-09:
Please don't post messages unrelated to the issue.

@hichemsystem1 you can reach me out on Home Assistant Discord https://www.home-assistant.io/join-chat I go by @dgomes

@Hadatko sorry, never followed up on this :/

Hadatko, 2024-11-09:
Sorry for spam. No problem, i didn't work on that since then.