dgomes/dali2mqtt #69: Sniffing data on Dali bus

Comiter, 2024-12-23:
Would it be possible to add functionality to sniff data on DALI bus, to get light status updated when a DALI switch or dimmer is used? This way I would not have to cyclically request DALI status for each DALI device, which take quite a lot of time.

This way, dali2mqtt would know the status of each switch based on either user input from a physical switch/dimmer, and by the commands sent through MQTT. When a dimmer/switch is utilized, this status could be sent through MQTT immediately (to for instance Home Assistant).

dgomes, 2024-12-25:
This daemon depends on python-dali library... I only support what is available through that library

Comiter, 2025-01-23:
I do believe that the python-dali library do support sniffing data, looking at the example async-buswatch.py in the python-dali repo?